KG-Development/SmartBMSUtility #1: Not able to differentiate between charching and discharchching state

nachtmaar, 2021-01-02:
It seems that for my battery (Liontron LifePo4 150Ah) the app cannot detect whether it is disgarching or charging.

The screenshot of the app was taken while charging via Solar, see second screenshot.

Let me know if you need more information.

[image]

[image]

NeariX67, 2021-01-02:
Ich seh schon. Da haut irgendwas nicht hin. Ich werde mir da mal in Sachen Frontend was ausdenken müssen. Ich arbeite zurzeit an der Configuration vom BMS, das hat erstmal als Version 1.1.0 vorrang. Ich denke das wird dann in Version 1.1.1 zusammen mit LifePo Support gefixt. Eine negative Leistung sollte eigentlich für entladung stehen, warum dann Peakpower bei 0.00 bleibt ist mir ein Rätsel. Ich teste das mal mit meinem Setup nachher.
----------English---------
Something isn't working right. I'm currently working on v1.1.0 that will allow bms configuration. I will fix this frontend issue in v1.1.1 together with LifePo support. Negative power should mean discharge, but the peakpower isn't updating for some reason. I'll take a look into it.

nachtmaar, 2021-01-02:
Thank you! Let me know if you need need me to test the fixed version (via TestFlight).

NeariX67, 2021-01-02:
Edit: Bei mir steht bei entladung ganz normal +80W und Peakpower geht auch. Hast du die andere BMS App installiert und kannst daraus mal ein Screenshot anhängen?
Seems to run normal for me. Both of my batteries report a positiv power while discharging. Maybe you accidently swapped your charging and discharging port of your BMS?
Could you try it with the App of Lukas Deszynski and see if it still shows that is it "charging"?
https://apps.apple.com/de/app/xiaoxiang-bms/id1375405426

NeariX67, 2021-01-02:
Just watched your issue again and noticed the correct behavior. I thought you are discharging and it reports charging but taking a look at „remaining chargetime“ reveals that it is indeed charging like intended. Negative current and power means charging, positive current and power means discharging.

closed 2021-01-02 by NeariX67

nachtmaar, 2021-01-02:
Sure, here are the screenshots.

If the battery is charging, it is getting energy, hence the sign should be positive. Then discharging, you take energy from battery, hence it should be negative sign, don't you think so? All other apps have the sign like that.
I checked, then discharging it has indeed positive sign.

[image][image]

NeariX67, 2021-01-03:
I'll invert the current reading and it will be available in version 1.1.0 (Coming soon™)

NeariX67, 2021-01-06:
Hey, if you are interested, you can test the early beta of v1.1.0. It currently features a configuration page that allows reading (and writing soon) of parameters that your bms uses to protect your cells.
I might need your apple id to invite you for testing.