KG-Development/SmartBMSUtility #5: Enhancement? Arduino API Library...

madmacks59, 2021-02-16:
The app you built is very handy and I use it regularly to monitor several batteries. But, I've been looking for a set of libraries/classes for the Arduino platform (or even RPI) that would allow me to read various parameters from the BMS using Arduino code. The use case for me is I have 5 different stand alone solar/battery sites at my off grid site. At each site I have a NodeMCU controller I've built that reads battery voltage and temperature and then connects to a remote MariaDB via WiFi and logs the data every 5 minutes. I'd love to have a library/class that let me connect to the BMS and retrieve state parameters directly from the BMS rather than using onboard sensors I have attached to the NodeBMS. And I think this would be useful for many other folks. NodeMCU and/or Arduino supports Bluetooth and WiFi connectivity so the missing piece is the class/library to query the BMS, and you have already figured out that part. And if you don't have the time to do that maybe you could point me to the code you used to retrieve the parameters in your Swift code and I could try to refactor it into Arduino C code and build the library myself.

NeariX67, 2021-02-16:
[This file](https://github.com/simat/BatteryMonitor/blob/master/BMSdecoded.xlsx) helped me a lot to understand eeprom addresses and what data to expect from them.
General communication can be found here: https://www.lithiumbatterypcb.com/smart-bms-software-download/

You could use the serial connection on your ESP8266 to communicate to it and save bms data to your DB

NeariX67, 2021-02-16:
But I have no Idea what my App should have to do with this?

closed 2021-02-17 by NeariX67

madmacks59, 2021-02-17:
Your app already has all the code to understand how to communicate with the BMS so my though was that the packages could be repackaged and deployed as an API library that could be used in other ways. That's how "your app has something to do with this". Not a bug, but a different way to leverage your excellent work...If I understood more about how to deal with the hardware I'd try myself. Maybe I'll give it a try as it sounds like a good learning opportunity...

NeariX67, 2021-02-17:
Running an API from an iPhone while beeing connected to a bluetooth device is comletely overkill. Any ESP32 with Bluetooth and Wifi can do it. Or just use the serial connection and transmit your data via wifi only. There are already quite a few arduino scetches written that communicate with the bms. Just push that to your DB and you are set.

Quick github search reveals libraries like [this](https://github.com/rahmaevao/JbdBms)

madmacks59, 2021-02-17:
Wasn't thinking of running an app on an iPhone...apparently I haven't been clear...

Your app is very nice and includes a nice GUI, processing logic, and a stable way to communicate between the code logic and the BMS hardware (typical MVC process I assume).

It is great for reading, displaying, and updating the BMS data. Well done.

All I was suggesting is that the BMS comms component might be leveraged to create a class/api that could be published as a stand alone class that could be used on different platforms like Arduino/NodeMCU/RPI.

If a library was available for the Arduino/NodeMCU/RPI to be able to communicate with the BMS via Bluetooth (or even UART) then people could read the BMS and do other kinds of things with the retrieved data, outside of the capabilities that you provide in your app. The use case I have is...

I live off grid and have several separate outbuildings (5) with their own solar/battery set ups. All these outbuildings are connected with microwave links to provide a common LAN. I have stand along NodeMCU devices monitoring all these batteries (temp, voltage, etc) and they all connect to a MariaDB via the LAN and log their collected data every 5 minutes. I also have a HTTP server running that reads all the data and presents me with a clean view off all the locations.

If I could use my Arduino/NodeMCU devices to directly communicate with the BMS via Bluetooth or UART then I could grab BMS info directly and would not need all the other sensors. The missing piece for me is a class library the allows that communication, and your code base has that already built.

Your app provides a lot more features than an Arduino solution would, with two way communication. And I can only access the BMS' when I'm in Bluetooth range via your app. So having the Arduino based ability to read and log the data to the DB would be great...and that's all I was suggesting.

Anyway...nice job, I really like the app you built. All the best...Gary

NeariX67, 2021-02-17:
I mean yes, I've build a rather large environment that allows the two-way communication, but it's nowhere near to be compatible with Arduino. I don't know whether I or you are confusing something here - I am not the manufacturer of the BMS or in any relation with that. I just have a rather large experience handling with these. And all I could teach you is already written down by others (see links I provided earlier). [This repository](https://github.com/rahmaevao/JbdBms) seems to be a great start to communicate with the bms and should have sufficient data available.