KG-Development/SmartBMSUtility #6: Can‘t reactivate BMS

LiontronUswe, 2021-03-08:
Hi,

I installed the app, connected to my Liontron 200ah and pressed the on/off buttons in the overview screen. The BMS turned power and charging off, and kept spinning afterwards. After restarting the app the buttons are green and when I press it it starts spinning but does not turn it back on the BMS. All power is off and in the Liontron App it says „Software deactivation BMS“. If I hit the on/off button again the your app, the waiting Symbol just keeps spinning but no bms. Please help.

NeariX67, 2021-03-08:
Hey, thanks for notifying me about this issue. I've been getting many issues around that On/Off Button problem, but i can't seem to fix this as of now. I hope i can finish the next version this weak and I will take a look into this issue as soon as I can.
Espacially Liontron batteries have a different Bms version that disallows reading and writing certain parameters. As I don't experience these issues with my own batteries (non-Liontron) I am unable to investigate this properly. But I'll see what I can do.

LiontronUswe, 2021-03-08:
Hi. I am just an average user, who owns a camper van and just bought a 1500€ battery which is now unusable. If this is not fixable you should at least put up a big disclaimer for Liontron battery’s. Everyone who hits one of these two buttons will not be able to use the battery until the issue is fixed. I will try to contact Liontron support. I hope they will be able to help me.

NeariX67, 2021-03-08:
You could also try to use this app: https://apps.apple.com/de/app/xiaoxiang-bms/id1375405426
This app should either have the same problems with Liontron Batteries or it should just work. This is not a serious and problematic issue you are dealing with

LiontronUswe, 2021-03-08:
Well to me it is. The battery is not giving out any power. Nor will the BMS allow to charge it. The battery is basically dead right now. How is this not serious?

NeariX67, 2021-03-08:
Are you using the MacOS version?

LiontronUswe, 2021-03-08:
I am using the iPhone App.

NeariX67, 2021-03-08:
I'd really try the other app until I can get this fixed. I hope I get done until Wednesday, so it will be on the appstore until friday.

LiontronUswe, 2021-03-08:
The App looks quite untidy. What exactly should I try to fix the issue?

NeariX67, 2021-03-08:
Yeah thats why I coded this app because the other app is just... different.

On the bottom left is a "batt on" button. Click it to enable your BMS

LiontronUswe, 2021-03-08:
Its 6,99€ to do that. Should it enable charging and discharging? I’ll try that later.

NeariX67, 2021-03-08:
Oh, didn't noticed it had the paywall behind it as i bought the app months earlier. Do you have an Android phone nearby? I don't own any Android device, but i know there's an official app from the BMS manufacturer that should allow you to turn the bms back on. Otherwise i'm out of ideas to support you

LiontronUswe, 2021-03-08:
No. There is no Android phone here. What is the app called? How can you leave that App in the appstore without a warning notice, knowing that the app has flaws that can render a battery useless. I will buy that app now if that may makes the electrics in my van functional again.

LiontronUswe, 2021-03-08:
Ok. The 6,99€ did the job. But dude. This needs to get fixed or to be pulled off the appstore. You are liable for the stuff you put into the AppStore.

NeariX67, 2021-03-08:
As I said in the App's description: "Liontron batteries have this kind of management system built in and are supported, yet untested."
I do not have 300€ laying around just to buy one for testing. I have to rely on people like you that report me these issues. And when I wrote the App's describtion, it was NOT known to me that Liontron offers limited functionality. I'll keep that in mind for the upcoming update.

NeariX67, 2021-03-08:
Glad you got it sorted.

LiontronUswe, 2021-03-08:
It is tested now. It breaks the battery in the current version when pushing that button. Your app appears in the appstore when searching Liontron. I am among these people who do not read the whole paperwork before trying something that is rated well.

LiontronUswe, 2021-03-08:
Thank you for helping me get it back running anyway.

NeariX67, 2021-03-08:
I've contacted Liontron to check if they are willing to send me a sample for the purpose of enhancing this app. The only other way I could get this going is by getting a hand on such a battery on another way. You are not living nearby Berlin and are willing to let me have it for a week? 😅

LiontronUswe, 2021-03-08:
It’s build into a camper van. Yes, I do live in Berlin but I do not want to pull it out. Ask greenakku.de if they can send you a test unit. I am sure they will. And PLEASE: Put up a big warning for Liontron users. Not everyone will find you on github. These battery’s are used for campaign a lot. Camping people are not knows for their IT capabilities.

NeariX67, 2021-03-08:
Yeah, you are completely right. I've started creating that info-screen right before you said it.

LiontronUswe, 2021-03-08:
You could walk along a campsite with this app and cause a lot of trouble due to the lack of PW protection in these BMS. I would appreciate if you would talk with liotron about that too.

NeariX67, 2021-03-08:
About password protection or about a test unit? I already requested such a unit from liontron and greenakku :)

NeariX67, 2021-03-08:
Walking through a campsite would allow me to discover all those BMS' but it wouldnt really benefit me. I just can't turn random peoples BMS off :c

LiontronUswe, 2021-03-08:
Morally? Practically it’s totally possible. People will be screwed if a random dick turns off discharging at their Liontron, but the app does not have the functionality to enable it again.

NeariX67, 2021-03-09:
Just checked the data that is transmitted by both my app and the "XiaoxiangBMS". They are exactly the same commands. It makes no sense to me that my App can't fix it, but his does.
LionTron answered my E-Mail this morning that they are proud about "this interesting request" and they moved my request to the corresponding department. I hope i'll get my hand on such a battery shortly.

closed 2021-03-11 by NeariX67

mbay0r, 2023-03-10:
Hello, does this bug still exist at the moment? thanks

NeariX67, 2023-03-10:
This "bug" can still happen if you have a password protected battery and that password hasn't been set in the app (version 2 or newer)

mbay0r, 2023-03-10:
by password protected battery, you must mean a password protected bms, right? So if I use the Overkill Solar to clear the BMS password, there should be no problems, right? thank you

NeariX67, 2023-03-10:
Should solve it, yes