KG-Development/SmartBMSUtility #9: iOS App Crashes when selecting Overview tab

usfshawn, 2021-04-02:
Design of app is excellent!

I am having an issue when clicking the overview tab. The overview tab was working correctly until I updated the total and cycle capacity in the config tab. I change the values from 100,000 to 280,000 and 80,000 to 220,000 respectively. These values did not seem to save in the BMS as the next read showed them at zero. I can still read/write on the config tab but the app crashes when I click the overview tab. 

Thank you and let me know if I can help with or test anything.

NeariX67, 2021-04-02:
Already solved it in my head. It is caused by the same issue as #8. i hope i get enough spare time to fix and upload this

usfshawn, 2021-04-02:
Thank you! Yes, I removed one 0 from the total capacity and it’s no longer crashing. Super app

closed 2021-04-02 by usfshawn

NeariX67, 2021-04-03:
Hmm.. App doesn't crash for me. I just wrote 210000 mAh to the BMS and it reports 0 mAh after reading. So far so good. But no crashes when i go back to the overview page

NeariX67, 2021-04-03:
Can you enable a specific feature in iOS for me? It will send the crash report to me (anonymously) and i can take a better look into that

iOS Settings -> Privacy -> Analysis and Enhancements (idk if it is called that way, my phone is in german) and then enable "share with ios developers"

Thanks

usfshawn, 2021-04-06:
Sure, I enabled the crash report sharing and the issue seems to be resolved with the latest build. Vielen Dank! Meine Frau kommt aus Deutschland.