KG-Development/SmartBMSUtility #18: Empty device list

RasmusHJV2, 2022-03-30:
When there is no devices in the device list, its pretty empty, idea could be to add some text saying "Couldn't find any devices in your area".

NeariX67, 2022-03-30:
I can do that in v2

RasmusHJV2, 2022-04-02:
Perfect! 😃

northlandb, 2022-04-07:
I just opened the app and now I am seeing nothing in the device list, I would like to add my 16S JBD BMS to the compatibility if possible?

What do you need from me?

NeariX67, 2022-04-07:
> I just opened the app and now I am seeing nothing in the device list, I would like to add my 16S JBD BMS to the compatibility if possible?
> 
> What do you need from me?

Thats weird. Could you please download the "BlueCap" App in the Apple App Store. Click Scan in the top left corner and search for your BMS in the device list and tap it. Click Services, take a screenshot and if "FF00 - Unknown" exists, tap that as well and screenshot that.

When did you buy this BMS?

closed 2023-01-09 by NeariX67