KG-Development/SmartBMSUtility #19: Unable to write on the BMS

atom-2jong, 2022-04-18:
[image]
[image]
I can't read/write on the BMS, the button is grayed out. I can't activate charge/discharge either. However, I can do it via the XiaoxiangBMS app.

katter, 2022-04-25:
Seeing the same behaviour on a JBD-SP04S028-L4S-60A-B. And config read/write works with XiaoxiangBMS, switching Charge/ Discharge works with neither one.

NeariX67, 2022-04-25:
Are you using a DIY battery? I haven't seen BMS' where my protection kicks in that are not from LionTron.
I'm gonna order that particular bms and test it out

katter, 2022-04-25:
Oh wait, it's a "Wipuvo" battery pack. Is there any way, I can do something to provide you with helpful data?

atom-2jong, 2022-04-25:
Yes, it's a DIY battery. The first time I launched the Smart BMS utility app, it worked, I could change the settings and save them as well as activate charging/discharge. But then I changed the BMS settings with the XiaoxiangBMS app and since then it no longer works with Smart BMS utility.

NeariX67, 2022-04-25:
What my guess is for both of your cases is that you used the xiaoxiang app to set a hardware password, which is not yet supported by my app. I tested the Xiaoxiang app to some extend and noticed some issues regarding the hardware pin, but he was unwilling to fix it.
https://github.com/smagicld/xiaoxiangBMS/issues/177
https://github.com/smagicld/xiaoxiangBMS/issues/176

So far - wait for version 2 of this app, currently working on it. You may test the beta of v2 with the testflight link in the description (WARNING: Missing important features so far)

atom-2jong, 2022-04-25:
Thanks ! But the beta is complete, I can't join it…

atom-2jong, 2022-04-25:
[image]

atom-2jong, 2022-04-25:
[image]

NeariX67, 2022-04-25:
Oh it probably expired already. I hope I can submit a new release today or in the next few days.

NeariX67, 2022-04-25:
Just uploaded a new beta of v2. Should be available through the invitation link.

atom-2jong, 2022-04-26:
Thanks you so much !!