KG-Development/SmartBMSUtility #21: Suggestion

atom-2jong, 2022-06-11:
It would be convenient to be able to easily set a maximum percentage, a maximum total voltage or a maximum voltage/per cell, in order to protect your battery and not leave it fully charged for a long time. The battery will charge up to a temporary and predefined "threshold", until we want to remove this threshold and charge it fully.!
[image](https://user-images.githubusercontent.com/81765688/173183112-895fb346-0b56-4bd1-825b-c8fe39db15d5.jpeg)
For this I would see a cursor (as in the photo) that would most of the time be left fully so that the battery charges fully, but that we could "reduce" to charge the battery up to a certain threshold

NeariX67, 2022-06-11:
Hey, thank you for your suggestion.
I see 3 possible ways to accomplish this:

- Manually change the cell or pack overvoltage protection to the desired voltage, so that the bms would automatically stop charging once the voltage limit is reached (would work with the current versions, no need for an update)
- The App monitors the voltage and sends a signal to the bms to disable the charging port (requires active bluetooth connection all the time)
- Create a dialog where you can select the charging limit and the app does the rest

Option 3 would be optimal in user perspective, i can take that onto my todo list for version 2 (You might want to join [TestFlight](https://testflight.apple.com/join/YWdbkZ8s))
But as long as the update isn't there, you can change the bms parameters to accomplish this

atomtoto, 2022-06-11:
Thank you for the answer! I was thinking instead of option 3 to have an interface a bit like on the tesla app. The app will remember the basic settings of the bms, and will change/restore them when moving a cursor or changing an option. I realize that it shouldn't be easy to code but it would be super useful! PS: I joined the TestFlight beta ;)

NeariX67, 2024-05-16:
Will be in v3.3.0

NeariX67, 2024-05-21:
Version will be named 3.2.5 instead of 3.3
We are currently preparing for release.

closed 2024-05-21 by NeariX67