KG-Development/SmartBMSUtility #23: Export Log as csv

ulfertg, 2022-11-03:
Hello,

I'm looking for a function to export the log as csv, but I can't find it.
Am I overlooking it or is it not there?
If I remember correctly an earlier version of this App provided this function and it was very useful to me.
If it's missing, would it be possible to implement it in  a future version?

NeariX67, 2022-11-03:
Yes you are right, version 1 had it. I can't find any time to do that in the near feature but i have it on my list.

NeariX67, 2023-01-08:
I have implemented a way to export data to csv like requested, still working on quality control and some other features before releasing a new version to the appstore!

ulfertg, 2023-01-08:
Great, thank you for the update!
If you have a Testflight Beta I'd be happy to help testing it.

NeariX67, 2023-01-08:
Did some more work on v2.1 and uploaded it to testflight. It still needs to be reviewed by apple, but you can expect a testflight version soon.
https://testflight.apple.com/join/YWdbkZ8s

ulfertg, 2023-01-09:
Thanks, I just gave it a go and it works great!
One minor issue I've encountered: I believe there are some redundant semicolons in the last columns (cell voltages).
Screenshot:
[image]

NeariX67, 2023-01-09:
Good catch. I'll take a look

ulfertg, 2023-01-09:
Also:  SOC Percent seems to be always 0 in the csv, while in the app's log pane it's reported as 100%.

The temp values would've been my next question, but you've already answered it :)
Personally, I'd prefer the temp scale to be the same as in the app - it's less confusing I guess.

NeariX67, 2023-01-09:
Fixed all the mentioned issues, now that the previous version got approved by apple it should be a matter of minutes until the next version hits testflight

ulfertg, 2023-01-09:
Updated the Beta, everything looks good now! Thanks!

Two thoughts about the values: 
1. Since the total Voltage is exported as Volts, it might my more consistent to export the cell voltage also as V instead of mV (e.g. 3.342 instead of 3342).
Also in the App the cell voltage is reported as V. Maybe there is a reason to report it as V which I'm not aware of.
2. The total Voltage seems to be reported by adding up the cell voltages and then truncating the last digit. I would've expected it to be rounded.
E.g. all four cells of a battery are at 3.342V, so the sum would be 13.368V, reported is 13.36V. I'd expected it to be reported as 13,37V. But this is very minor and of no practical consequence I could think of.

Thanks again for bringing the export feature back, to me it's very helpful for testing batteries and balancers.

NeariX67, 2023-01-09:
The total voltage is provided by the bms, I could indeed calculate it myself, but i don't like handling it that way.

I tried to avoid exporting cell voltages in volts because it would mean that there are commas and dots, which might causes some problems in a comma separated file.

I will have some thoughts about that

closed 2023-03-01 by NeariX67