KG-Development/SmartBMSUtility #24: MAC address Display

madmacks59, 2023-01-09:
I have many BMS' and they all have names stored on them. When I launch the app I see a list of generic names which makes it difficult to connect to a target BMS. And the MAC addresses displayed on the tiles is microscopic so unusable to select.
[image]

NeariX67, 2023-01-09:
Yeah you're right, its unreadable and I did that intentionally for whatever reason. But I've not yet seen how to read the actual MAC-Address instead of using a random UUID (Identifier). I will change that, but don't know how right now

madmacks59, 2023-01-09:
I wish I had the knowledge to help. My comments are just feedback and not criticism! I haven't done Swift programming but it's on my list of things to learn. I do do (ha I said Do Do) Arduino programming and am trying to build a device that sits connected all the time to a BMS using UART, reads various values, and then remotely logs that data to a MariaDB table on one of my Raspberry PIs. I have 5 separate battery systems and I want a central web page to view all their conditions in one place. Figuring out the API to read the values on the Arduino is a PITA...so I appreciate your efforts!

madmacks59, 2023-01-09:
Yep, they show up...

[image]

NeariX67, 2023-01-09:
Huh. They show up with the "wrong" names. So the other app stores the device name somewhere else, which I can't read and thus the other name shows up. My App works as expected in that case.

closed 2023-01-09 by NeariX67

madmacks59, 2023-01-09:
Looks like the "other all" show three places where name is used.
[image]

And...

[image]

"BMS Alias" appears to be app specific (ie not on the BMS).
"BMS Bluetooth Name" I'd think is on the BMS and should be available to applications.
"PIN/BMS Name" appears to be what your app is pulling. Maybe you can access the "BMS Bluetooth Name" as the "PIN" version, as I understand the manufacturers info should not be changed.

NeariX67, 2023-01-09:
Even though the name is properly set in BMS Bluetooth name, the BLE Scanner still shows "xiaoxiang BMS" and therefore that's the name my app displays. Try setting the name(s) again with the other app.

NeariX67, 2023-01-09:
Btw, you can also set a custom name for your BMS' in the settings of a device.
[image]