KG-Development/SmartBMSUtility #25: Not reading the configured battery size.

madmacks59, 2023-01-09:
When I select what I think is the target BMS (see other issue I raised) the app displays an incorrect value for the AHs if the battery.

Display from factory app...
[image]

Display from this app...
[image]

NeariX67, 2023-01-09:
Seeing that the capacity values are below zero, i think the capacity is larger than what can be stored in whatever variable type i used. I hope using a better variable type would fix that. I'll make some tests and release a fix.

NeariX67, 2023-01-09:
Can you download 'BLE Scanner' from the appstore and tell me whether your BMS' are showing up with the correct names?
[image]

NeariX67, 2023-01-09:
Fixed. New testflight version very soon.

closed 2023-01-09 by NeariX67