KG-Development/SmartBMSUtility #26: Suggestion: create possibility to adjust y-axis in logging view

ulfertg, 2023-03-13:
Hello,
it would be awesome if the y-axis of the charts in the logging view could be adjusted manually, for example in the same way as the x-axis with plus and minus buttons, or maybe even more intuitively with pinch gestures.
Alternatively and maybe easier to implement, an automatic zoom to set the min y value of the chart a little bit lower than the value of the lowest reading in the view (I believe the same is already happening with the max y value, for example in the temperature view).
I think this would make the information displayed there much more accessible.
Especially info like the individual cell voltages, where deviations of a couple of mV could be of interest, but are not visible in the chart at the moment because it's projected to a range of 4000mV.

In the past I just exported the log as csv and then used a spreadsheet to generate the charts, but it would be really handy to be able to take a quick glance at the app and see how the readings change over time.

NeariX67, 2023-03-13:
I completely agree with your suggestions, but i have not found a decent way to implement any of this. But once Apple updates their Chart library i'm happy to take this to action!

NeariX67, 2023-03-13:
Huh, actually found some decent examples on how to do this. I'll take a look into it!

ulfertg, 2023-03-13:
That's great!
I was just writing a response which is now probably obsolete, but I leave it below for your consideration:

Thank you for the information.
I don't know it that helps, but I thinks one of the most intuitively ways to implement a responsive chart can be found in the app SensorPush.
Also Kai Scheffer implemented resizable charts in his app BlueBattery.

NeariX67, 2023-08-04:
Implemented in V3

closed 2023-08-04 by NeariX67