KG-Development/SmartBMSUtility #31: Cannot click Connect button when Dynamic Type text size too large

RBL66, 2023-08-27:
Hello,

Thanks for the great app!

I have found that when using your iPhone app, if I set Dynamic Type fairly large, the “Connect” button moves *outside* of the box showing the battery name. Then the Connect button is not responsive to any taps so the battery screen cannot be opened. The only workaround is to go to iPhone Settings and change Dynamic Type a little smaller to let the Connect button move inside the battery name box. 

Just thought I’d report this little bug. It was hard to find because it looked like the button was just not responding on one phone and working on the other. But we couldn’t see why. 

Thanks!

Robert Burfoot-Lobo

ImVelas, 2023-08-27:
Hey @RBL66,
thanks for reaching out to us. We noticed the problem and started working on it.
We´ll release the v3.0.1 shortly and it will contain a fix for it.
Best regards
Fabian

NeariX67, 2023-08-29:
Update 3.0.1 is uploaded, release following shortly.

closed 2023-08-29 by NeariX67