KG-Development/SmartBMSUtility #32: Behaviour of the port switches is counterintuitive on newer Versions of JBD BMS

ulfertg, 2023-08-29:
Hello

Newer JBD BMS apparently automatically open a closed port if a certain current (in my case it's 2A) is exceeded. This prevents heating of the Mosfets (because otherwise the current needs to travel through the body diode).
I cannot test it right now, but I believe older versions of JBD BMS did not behave like that.

Anyhow, this makes the behaviour of the Charging / Discharging Buttons hard to understand.
I hope this examples will help clarifying:

- The battery is being charged with 1A.
- The button "Discharging" is being triggered
- The button turns red, the discharging Mosfets are being locked, the outer voltage rises about 500mV, because the charging current needs to travel through the body diodes of the discharging Mosfets.

So far, everything as expected in this scenario - but now the charging current rises to 3A and then this happens:

- The BMS unlocks the discharging Mosfets, the "Discharging" button turns blue again (respectively to the selected theme's color which reflects the open port state)
- It's not possible to unlock the discharging port in this state, pressing the "Discharge" button seems to fail to alter the state.
- Only when the current drops below 2A, the button turns red again and it's possible to unlock the port.

Even more confusing: 

- The battery is being charged with 3A
- The button "Discharging" is being triggered
- It seems to fail to alter the state.
- After the charging current drops below 2A, the discharge port is being locked.

While those example are referring to discharge, this happens also in the other direction.

So, in newer JBS BMS the switches are rather a request than a trigger.
And this request can be overridden by the BMS itself, but then the buttons won't reflect it.

This is how I would expect the buttons to behave:
- The buttons don't show the actual state of the ports, but the request
- Maybe the actual state of the ports is being displayed too, in my opinion this would be a "nice to have", but not a necessity.
- The request can be changed while it's being overridden by the BMS.

Kind regards
Ulf

PS: I like the new V3 a lot!

NeariX67, 2023-08-29:
Thank you for your issues and feedback.
My guess is that this issue is more of a BMS firmware issue, I suspect that we can't do much about this here.
But I will test and confirm this and take action accordingly.

ulfertg, 2023-08-29:
Thanks for the quick reply!

I tested a couple of different apps and while they all show inconsistent behaviour of the port buttons, some of them show the actual state of the Mosfets when locked (XiaoXiangBMS & XiaoXiangElectric, though they don't show the specific port, rather they report a generic Mosfet software lock) and also can revoke the lock request while the Mosfets are temporarily unlocked due to current above 2A.
So the BMS firmware should provide the function and information.
Hope that helps.

NeariX67, 2023-11-20:
@ulfertg Can you tell us your BMS Firmware version of the effected BMS?

ulfertg, 2023-11-20:
@NeariX67 The BMS Firmware Versions is being reported as "20".

NeariX67, 2023-11-20:
0x20 or "20"? Because the v3.0.5 displays the version in hexadecimal format now.

ulfertg, 2023-11-20:
Indeed, your App reports it as "0x20", XiaoxiangBMS app reports it as "20".