KG-Development/SmartBMSUtility #34: Pinch zoom of the diagrams in the logging view

ulfertg, 2023-08-29:
Hello

I really like the new logging view, it's a big improvement over the previous diagrams.

I understand that it's possible to narrow the viewport by selecting a certain time and date.

But it would be awesome if it would be possible to add pinch zoom functionality to the diagram view.

This would make navigation through the data points more intuitive in my opinion.

Kind regards
Ulf

NeariX67, 2023-11-20:
Waiting for this issue to get resolved:

- [ ] https://github.com/imaNNeo/fl_chart/issues/71

NeariX67, 2024-11-26:
There is now some 'motion' in the fl_chart library. Some guy is willing to put some effort into this, so having this feature in our app is coming closer!

NeariX67, 2024-12-20:
This feature got implemented in [fl_chart](https://github.com/imaNNeo/fl_chart/)
We need to do some performance optimization in our logging view and we might be ready to release v3.5.0!

closed 2024-12-20 by NeariX67